englmaxi/zmk-dongle-display #2: modification to use vertical  display

radsonpatrick, 2024-11-11:
I am using a 128x32 OLED display, and I would like to rotate the icons displayed to improve the layout and arrangement on the screen. Currently, the icons are not optimally oriented for this display size and orientation.
[image]

radsonpatrick, 2024-11-11:
my intencion is rotaete and realocate objects ! 
example rotate icon bluetooth im try add this functions `lv_img_set_angle(bt,900);`  or ` lv_obj_set_style_transform_angle(bt, 900, 0);` in `./boards/shields/dongle_display/widgets/output_status.c` 

but after compile  the icon is blank !

you help-me ?

englmaxi, 2024-11-12:
It seems that there is no easy way to rotate the entire screen: https://github.com/zmkfirmware/zmk/issues/1749

You can use the mentioned workaround or, instead of rotating the images via LVGL, try to rotate the original images.

<details>
  <summary>original image files</summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

</details>

closed 2024-11-15 by englmaxi