englmaxi/zmk-dongle-display #4: Mac Icons are default and I am unable to enable windows icons.

Goose-Dev, 2025-02-16:
Mac icons are default despite not specifying:
ZMK_DONGLE_DISPLAY_MAC_MODIFIERS=y

I've tries specifying n as well as y and nothing changes it.
 This probably is user error but any help is appreciated.

englmaxi, 2025-02-16:
Hi,

you have copied a version of `dongle_display` that is not up-to-date with the latest version of `main` into your `boards/shields` directory. I would recommend deleting the folder completely. Then the current code is automatically loaded from this repository and any further improvements/bug fixes will be available for you.

Let me know if this helps or if you have any other questions :)

Goose-Dev, 2025-02-16:
That was it. Thanks much for the help!

closed 2025-02-16 by Goose-Dev