englmaxi/zmk-dongle-display #5: layer_status conflitct

redmasters, 2025-02-17:
Hello, first of all. Thank you for making the project publicly available; I learned a lot by implementing it in my own setup.
I forked your project and am using it as a module. During my implementation, I encountered an issue with conflicts in the layer_status. To resolve this, I removed all references to layer_status from the widgets, and it worked perfectly. 
Below is my current configuration: 

[image]

Now, I’d like your help, if possible, on how to properly implement the layer_status—ideally displaying it below the Bongo Cat, centered at the bottom. 

Thank you in advance! Best regards!

englmaxi, 2025-02-22:
Hi, I'm glad to hear that you like the project!

I'm struggling to understand what exactly your problem is or what conflict you have observed. Can you describe your problem in more detail?

How you position objects with LVGL is described here: https://docs.lvgl.io/8.0/overview/coords.html#position

Unfortunately, I designed the widgets for a 128x64 pixel display, so there will most likely be overlapping on a smaller display.

englmaxi, 2025-04-19:
I am really sorry @redmasters that I am only getting back to you now...

The problem is caused by the configuration option `CONFIG_ZMK_WIDGET_LAYER_STATUS=y` in your zmk-config. This causes a conflict with the default widget. If you remove this option, it should work.

redmasters, 2025-04-19:
Thx @englmaxi , I'll try it

englmaxi, 2025-04-27:
I am closing this issue. You are welcome to reopen it if you are still having problems. :)

closed 2025-04-27 by englmaxi