englmaxi/zmk-dongle-display #7: cannot manage to show layer and two battery widgets

harmy-fgc, 2025-03-26:
i managed to get the display running, i got the cat, i got the modifiers, i got the battery widget for one half 

but i cannot manage to get battery information from both halves, and i cannot manage to get the information for the top layer. 

Any ideas? i don't know if it's a me - issue, or something with the software 

using a ssd1306 screen, and a nrf52840 Pro Micro 

https://github.com/harmy-fgc/harmy_zmk

harmy-fgc, 2025-03-26:
nevermind, it was a me-issue, love the module

harmy-fgc, 2025-03-27:
battery only for one half still, but tbh, i would rather assume that's a me-issue, or at least definitely not the module

i really appreciate the module, the cat makes my day lmao

englmaxi, 2025-03-31:
related to #8

closed 2025-04-07 by englmaxi

theharmy, 2025-05-19:
a very very late reaction, but yes, it works now, and it works great!