englmaxi/zmk-dongle-display #8: Missing peripherals battery levels when peripherals count >= 2?

linrongbin16, 2025-03-31:
When the keyboard has 1 dongle + 2 peripherals, the battery levels are wrong. It only shows 1 battery levels, missing another one.

In this line: https://github.com/englmaxi/zmk-dongle-display/blob/5dbeb0d03d26648d5ef454b46422e957ddfc2c32/boards/shields/dongle_display/widgets/battery_status.c#L37 

Your source code uses `ZMK_SPLIT_BLE_PERIPHERAL_COUNT`, but in zmk official document, the peripherals count is defined by: `CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS=2`.

See: https://zmk.dev/docs/config/split#bluetooth-splits .

linrongbin16, 2025-03-31:
Update:

After some investigation & testing, I found there are several configs may affected the battery level displays.

In my use case, my keyboard is a 1 dongle + 2 peripherals splits, connected by BLE. 

I wonder how should I configure these options:

## For Dongle (Central)
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS=2`
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_BATTERY_LEVEL_FETCHING=y`
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_BATTERY_LEVEL_PROXY=y`

## For Both Left and Right
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS=2`
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_BATTERY_LEVEL_FETCHING=y`
- `CONFIG_ZMK_SPLIT_BLE_CENTRAL_BATTERY_LEVEL_PROXY=y`

For now the dongle-display looks like:

[image]

It only shows 1 battery level, but both two peripherals are working correctly for me.

linrongbin16, 2025-03-31:
Similar issue: https://github.com/mctechnology17/zmk-dongle-display-view/issues/1

linrongbin16, 2025-03-31:
I fixed this issue by setting zmkfirmware to latest stable version `v0.2.1`, see: https://github.com/linrongbin16/zmk-cygnus-tokio2006/pull/12/files.

Guess there are some bugs or break changes.

englmaxi, 2025-03-31:
Thank you for your investigation! I was not aware of this. I'll reopen this issue and have a look at what has changed since the last stable release.

englmaxi, 2025-03-31:
It seems like there is a bug in zmk: https://github.com/zmkfirmware/zmk/pull/2901

englmaxi, 2025-04-04:
the PR was merged, it should work again. 🎉

linrongbin16, 2025-04-04:
> the PR was merged, it should work again. 🎉

Thank you!

closed 2025-04-04 by linrongbin16