englmaxi/zmk-dongle-display #9: Battery Level Display Issue When Powering On Without USB Connection

te9no, 2025-04-14:
When powered on while connected to USB power, the battery level is displayed. 
However, when powered on without any connection, the battery level is not displayed. 
If you know how to resolve this issue, I would appreciate your guidance. 
The code is as follows.
https://github.com/te9no/zmk-config-Solkatstice/tree/test

englmaxi, 2025-04-14:
Hi,

I was able to reproduce your issue by disabling both peripherals of my split keyboard. It is therefore not a configuration problem...

I will look at how we can solve this.

englmaxi, 2025-04-17:
Thank you for the investigation! It has pointed me in the right direction.

I fixed the problem by bringing the objects to the foreground when the hidden flag is cleared.

Could you try #10?

te9no, 2025-04-19:
Thank you for your prompt response.  
I tried it, and the issue has been successfully resolved. 
This brings my keyboard one step closer to completion!  
I truly appreciate it.  
Thanks again!

closed 2025-04-19 by te9no