englmaxi/zmk-dongle-display #12: Flexible widget width for long layer names

linrongbin16, 2025-05-06:
I set long names for layers:

- "Qwer-Mac"
- "Qwer-Win"
- "Media-Mac"
- "Media-Win"
- ...

Here's a picture of how it shows in the screen:

[image]

You can see the text is been truncated, I hope we can have a more compact text rendering when the string is too long.

englmaxi, 2025-05-11:
Instead of changing the font size, there is also the option for a scrolling animation:

[image][image]

Or do you think a smaller font would be preferable?

linrongbin16, 2025-05-11:
The scrolling looks great! But could you tell me how to config it? I didn't find it in readme.

englmaxi, 2025-05-12:
You can try it out by changing the revision in `west.yml` to `long-labels`.

Let me know if it works for you or if you notice any downsides.

closed 2025-05-12 by linrongbin16

englmaxi, 2025-05-14:
Have you had time to test this? If it works correctly, I'll merge it to main.

linrongbin16, 2025-05-14:
> Have you had time to test this? If it works correctly, I'll merge it to main.

I will try today, then reply you

PR #13 (2025-05-20): Enable scrolling on labels
Fixes #12