englmaxi/zmk-dongle-display #22: Battery widget for ESB connection?

kumbung01, 2025-08-19:
hello, thank you for this repo, 

I'm currently using ESB for connection, which is capable of reporting battery level.

however, the dependency for BLE prevents from initializing battery widget.

Can you remove dependency for BLE with battery widget?

you just have to remove `IS_ENABLED(CONFIG_ZMK_BLE)`, and it works fine.

thank you.

the code is in custom_status_screen.c, line 76.
`#if IS_ENABLED(CONFIG_ZMK_BATTERY) && IS_ENABLED(CONFIG_ZMK_BLE)
    zmk_widget_dongle_battery_status_init(&dongle_battery_status_widget, screen);
    lv_obj_align(zmk_widget_dongle_battery_status_obj(&dongle_battery_status_widget), LV_ALIGN_TOP_RIGHT, 0, 0);
#endif`

closed 2025-08-19 by kumbung01

kumbung01, 2025-08-20:
the reason was 
```c++
#ifndef ZMK_SPLIT_BLE_PERIPHERAL_COUNT
#  define ZMK_SPLIT_BLE_PERIPHERAL_COUNT 0
#endif
``` 
which 'needs' to be BLE on, by
```c
#define ZMK_BLE_IS_CENTRAL                                                                         \
    (IS_ENABLED(CONFIG_ZMK_SPLIT_BLE) && IS_ENABLED(CONFIG_ZMK_SPLIT_ROLE_CENTRAL))

#if ZMK_BLE_IS_CENTRAL
#define ZMK_BLE_PROFILE_COUNT (CONFIG_BT_MAX_PAIRED - CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS)
#define ZMK_SPLIT_BLE_PERIPHERAL_COUNT CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS
#else
#define ZMK_BLE_PROFILE_COUNT CONFIG_BT_MAX_PAIRED
#endif
```
SPLIT_BLE was not on, so no ZMK_SPLIT_BLE_PERIPHERAL_COUNT 

so guess I just have to fork your code and define it by myself...

badjeff, 2025-08-24:
You could try `CONFIG_COMPILER_OPT` to add back the macro derivatives guarded by `#if ZMK_BLE_IS_CENTRAL`. 
Or, appends CC FLAG via west build with `-D`

```
CONFIG_ZMK_SPLIT_BLE_CENTRAL_PERIPHERALS=2
CONFIG_COMPILER_OPT="-DZMK_SPLIT_BLE_PERIPHERAL_COUNT=2"
```

englmaxi, 2026-03-17:
> It was just one of the many things I tried to get it working. They are nicenano v2 clones. Not sure if that makes a difference. 

I would try removing the node and changing it to `nice_nano_v2`. You would need to reflash both halves.